ros-navigation/navigation2_tutorials #13: Pure pursuit controller not steering

Rayman, 2020-11-06:
Hi,

I'm following the navigation2 tutorial and and was testing the pure pursuit controller. It seems to me that the controllor is not steering. See the gif below.

#### Possible cause

The path is transformed only once (at `setPath`). This means the local goal is never updated. I think the path should be transformed at every `computeVelocityCommands` call.

[image]

SteveMacenski, 2020-11-06:
A PR to resolve would be appreciated. Keep in mind this was just tutorial code so I didn't really review it for functionality. I agree though that having a functional pure-pursuit controller would be helpful though for some users.

Rayman, 2020-11-06:
I've fixed the 2 issues I found, I'll clean it up for a PR soon™

PR #14 (2020-11-23): Fix 2 bugs in the pure_pursuit_controller that prevented it from working and add path pruning
1. The path should be transformed on every call to
   `computeVelocityCommands`
2. The local goal should be determined from the closest point to the
   robot onwards
3. Add path pruning

Fix #13

I've even got a gif:
[image]
commit 12690245 (2020-11-23): Fix 2 bugs in the pure_pursuit_controller that prevented it from working and add path pruning (#14)

* Fix 2 bugs in the pure_pursuit_controller that prevented it from working

1. The path should be transformed on every call to
   `computeVelocityCommands`
2. The local goal should be determined from the closest point to the
   robot onwards

Fix #13

* Uncrustify & fix most ament_cpplint issues

* Add path pruning inspired by dwb_core

* Add a small README to describe the design of the controller

* Make transformPose const

* Address review comments by clarifying comments and rearranging code [... cut]
closed 2020-11-23 by SteveMacenski