ros-navigation/navigation2_tutorials #20: error building nav2_pure_pursuit_controller

diogopbrandao, 2021-01-15:
Have been tasked with creating a wrapper to use a specific controller on ros2 navigation stack.
Came across this tutorial that has been really helpfull on understanding how it works:
[https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2controller_plugin.html](url)

As a first step I'm trying to test this nav2_pure_pursuit_controller example, but can't get pass building.

**Steps to reproduce:**

-  On navigation2_tutorials workspace run : `colcon build --packages-select nav2`

**Output**:

```bash
 ~/ros/navigation2_tutorials$ colcon build --packages-select nav2_pure_pursuit_controller
Starting >>> nav2_pure_pursuit_controller
--- stderr: nav2_pure_pursuit_controller                               
In file included from /opt/ros/foxy/include/class_loader/class_loader_core.hpp:57,
                 from /opt/ros/foxy/include/class_loader/class_loader.hpp:55,
                 from /opt/ros/foxy/include/class_loader/multi_library_class_loader.hpp:52,
                 from /opt/ros/foxy/include/pluginlib/class_loader.hpp:58,
                 from /opt/ros/foxy/include/nav2_costmap_2d/costmap_2d_ros.hpp:53,
                 from /opt/ros/foxy/include/nav2_core/controller.hpp:42,
                 from /home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/include/nav2_pure_pursuit_controller/pure_pursuit_controller.hpp:15,
                 from /home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/src/pure_pursuit_controller.cpp:14:
/opt/ros/foxy/include/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = nav2_pure_pursuit_controller::PurePursuitController; B = nav2_core::Controller]’:
/opt/ros/foxy/include/class_loader/meta_object.hpp:190:7:   required from here
/opt/ros/foxy/include/class_loader/meta_object.hpp:192:12: error: invalid new-expression of abstract class type ‘nav2_pure_pursuit_controller::PurePursuitController’
  192 |     return new C;
      |            ^~~~~
In file included from /home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/src/pure_pursuit_controller.cpp:14:
/home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/include/nav2_pure_pursuit_controller/pure_pursuit_controller.hpp:23:7: note:   because the following virtual functions are pure within ‘nav2_pure_pursuit_controller::PurePursuitController’:
   23 | class PurePursuitController : public nav2_core::Controller
      |       ^~~~~~~~~~~~~~~~~~~~~
In file included from /home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/include/nav2_pure_pursuit_controller/pure_pursuit_controller.hpp:15,
                 from /home/diogoesa/ros/navigation2_tutorials/src/navigation2_tutorials/nav2_pure_pursuit_controller/src/pure_pursuit_controller.cpp:14:
/opt/ros/foxy/include/nav2_core/controller.hpp:74:16: note:     ‘virtual void nav2_core::Controller::configure(const SharedPtr&, std::string, const std::shared_ptr<tf2_ros::Buffer>&, const std::shared_ptr<nav2_costmap_2d::Costmap2DROS>&)’
   74 |   virtual void configure(
      |                ^~~~~~~~~
make[2]: *** [CMakeFiles/nav2_pure_pursuit_controller.dir/build.make:63: CMakeFiles/nav2_pure_pursuit_controller.dir/src/pure_pursuit_controller.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:78: CMakeFiles/nav2_pure_pursuit_controller.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_pure_pursuit_controller [23.7s, exited with code 2]

Summary: 0 packages finished [24.1s]
  1 package failed: nav2_pure_pursuit_controller
  1 package had stderr output: nav2_pure_pursuit_controller 
```

SteveMacenski, 2021-01-15:
The controller API has changed so this tutorial will work on foxy but not master branch. 

We’re also migrating a derivative of this work into the main stack in the next few weeks (with a number of new features / improvements) so that should be available for use not too long from now. 

If you want this controller for tutorials, then just know its not your fault the interfaces have changed. If you want it to run on a robot, then if you wait a bit I think you’ll be happy you did.

SteveMacenski, 2021-01-22:
https://github.com/ros-planning/navigation2/pull/2152#pullrequestreview-574395605

Moved into the main stack - ready for use!

closed 2021-01-22 by SteveMacenski