ros-navigation/navigation2_tutorials #23: Declaring the planner plugin in the nav2_params.yaml (Foxy)

philison, 2021-01-31:
I am trying to use the _straightline_planner_ plugin in ROS2 Foxy. I placed the _straightline_planner_ in a separate workspace (not the main _navigation2\_ws_) 
Building the package is successful. 
I followed the tutorial and modified the _nav2\_params.yaml_ file in my _navigation2_ws_ to use the _straigline_planner_ as the global planner. Unfortunately when I start the example with: 
`$ ros2 launch nav2_bringup tb3_simulation_launch.py`
there seems to be a problem with loading the new planner plugin. 

I am not sure if the planner is not properly loaded or if the problem is with the _straighline\_planner_ itself (or the location of the package)? I also tried different variants of declaring the planner plugin in the yaml file. The simulation works, except that it is not possible to generate a path to the goal position, as the following output shows:

Thank you very much for your time ! 

Tutorial Code for the nav2\_params.yaml file:
planner_server:
`ros__parameters:
  planner_plugin_types: ["nav2_straightline_planner/StraightLine"] # For Foxy and earlier
  planner_plugin_ids: ["GridBased"] # For Foxy and earlier
  plugins: ["GridBased"] # For Galactic and later
  use_sim_time: True
  GridBased:
    plugin: nav2_straightline_planner/StraightLine # For Galactic and later
    interpolation_resolution: 0.1`

Using the suggested lines for Foxie and commenting out the lines for Galactic and later is not working at all, as this will lead to the usage of the default _nav2\_navfn\_planner_, completely ignoring the _straighline\_planner_. 

`[lifecycle_manager-13] [INFO] [1612121985.772204302] [lifecycle_manager_navigation]: Managed nodes are active
[rviz2-4] [INFO] [1612121985.773500244] [rviz2]: Waiting for the lifecycle_manager_localization/is_active service...
[rviz2-4] [INFO] [1612121985.773628489] [rviz2]: Sending lifecycle_manager_localization/is_active request
[rviz2-4] [INFO] [1612121985.796956654] [rviz2]: Trying to create a map of size 384 x 384 using 1 swatches
[rviz2-4] Start navigation
[bt_navigator-11] [INFO] [1612121987.372087209] [bt_navigator]: Begin navigating from current location to (-1.61, 1.47)
[planner_server-9] [ERROR] [1612121987.374854305] [planner_server]: planner GridBased is not a valid planner. Planner names are: GridBased 
[planner_server-9] [WARN] [1612121987.374901048] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.61, 1.47)
[planner_server-9] [WARN] [1612121987.374919063] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[planner_server-9] [INFO] [1612121987.394338651] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[planner_server-9] [ERROR] [1612121987.395192696] [planner_server]: planner GridBased is not a valid planner. Planner names are: GridBased 
[planner_server-9] [WARN] [1612121987.395233277] [planner_server]: Planning algorithm GridBased failed to generate a valid path to (-1.61, 1.47)
[planner_server-9] [WARN] [1612121987.395249987] [planner_server_rclcpp_node]: [compute_path_to_pose] [ActionServer] Aborting handle.
[controller_server-8] [INFO] [1612121987.404362275] [local_costmap.local_costmap]: Received request to clear entirely the local_costmap
[planner_server-9] [INFO] [1612121987.405222963] [global_costmap.global_costmap]: Received request to clear entirely the global_costmap
[recoveries_server-10] [INFO] [1612121987.405698346] [recoveries_server]: Attempting spin
[recoveries_server-10] [INFO] [1612121987.406445149] [recoveries_server]: Turning 1.57 for spin recovery.
[recoveries_server-10] [INFO] [1612121988.406905943] [recoveries_server]: spin running...
[recoveries_server-10] [INFO] [1612121989.206973602] [recoveries_server]: spin completed successfully
[recoveries_server-10] [INFO] [1612121989.234869369] [recoveries_server]: Attempting wait
[recoveries_server-10] [INFO] [1612121990.235019975] [recoveries_server]: wait running...
[recoveries_server-10] [INFO] [1612121991.234981786] [recoveries_server]: wait running...
[recoveries_server-10] [INFO] [1612121992.234988247] [recoveries_server]: wait running...
[rviz2-4] [INFO] [1612121992.764902519] [rviz2]: Message Filter dropping message: frame 'odom' at time 3,808 for reason 'Unknown'
[rviz2-4] [INFO] [1612121992.969423367] [rviz2]: Message Filter dropping message: frame 'odom' at time 4,009 for reason 'Unknown'
[rviz2-4] [INFO] [1612121993.173612944] [rviz2]: Message Filter dropping message: frame 'odom' at time 4,210 for reason 'Unknown'
[recoveries_server-10] [INFO] [1612121993.234997738] [recoveries_server]: wait running...
[rviz2-4] [INFO] [1612121993.378075367] [rviz2]: Message Filter dropping message: frame 'odom' at time 4,410 for reason 'Unknown'
[rviz2-4] [INFO] [1612121993.548095092] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 3,801 for reason 'Unknown'
[rviz2-4] [INFO] [1612121993.549621195] [rviz2]: Message Filter dropping message: frame 'odom' at time 4,611 for reason 'Unknown'
[rviz2-4] [INFO] [1612121993.752391465] [rviz2]: Message Filter dropping message: frame 'base_scan' at time 4,001 for reason 'Unknown'
`

SteveMacenski, 2021-02-01:
>  there seems to be a problem with loading the new planner plugin.

Is the other workspace sourced? that could be an issue if its not even in the same path or knows about each other.

> planner GridBased is not a valid planner. Planner names are: GridBased 

That looks interesting. Are you sure this even loaded correctly at all? Go back in your logs to the configuration stage for the planner server and make sure the planner even loaded at all. 

https://github.com/ros-planning/navigation2/blob/2efdad8eb838d28ab9f3c24fc4aa5136b0ae9bdc/nav2_planner/src/planner_server.cpp#L309

philison, 2021-02-01:
Thanks @SteveMacenski . It works now !
Although I failed to comprehend how I can set the severity level for loggers in Foxy, I followed your advice and took a closer look at the log file. You were right, the planner\_server failed to find the library. 

```
[planner_server-9] [FATAL] [1612217551.476260489] [planner_server]: Failed to create global planner. Exception: Could not find library corresponding to plugin nav2_straightline_planner/StraightLine. Make sure the plugin description XML file has the correct name of the library and that the library actually exists.
[planner_server-9] [INFO] [1612217551.476293479] [planner_server]: Planner Server has GridBased  planners available.
```

So I made sure that all the necessary ws are sourced and that the plugin declaration in the _nav2\_params.yaml_ is correct and it finally worked. I don’t know what I did different this time, which is not very satisfying as I spend far too much time on it :D But I am glad that it works now.

Thank you very much for your help and the hint ! I very much appreciate that you took the time and replied so quickly !   

Below is my final version of the _nav2\_params.yaml_ file. 

``` 
planner_server:
  ros__parameters:
    expected_planner_frequency: 20.0
    use_sim_time: True
    planner_plugins: ['GridBased']
    # GridBased:
    #   plugin: 'nav2_navfn_planner/NavfnPlanner'
    #   tolerance: 0.5
    #   use_astar: false
    #   allow_unknown: true

    planner_plugin_types: ['nav2_straightline_planner::StraightLine'] # For Foxy and earlier
    planner_plugin_ids: ['GridBased'] # For Foxy and earlier
    plugins: ['GridBased'] # For Galactic and later
    use_sim_time: True
    GridBased:
      plugin: nav2_straightline_planner/StraightLine # For Galactic and later
      interpolation_resolution: 0.1
```

closed 2021-02-01 by philison

root-robot, 2021-04-28:
> Thanks @SteveMacenski . It works now !
> Although I failed to comprehend how I can set the severity level for loggers in Foxy, I followed your advice and took a closer look at the log file. You were [...]

Hey @philison 

Can you kindly provide me with the link to your code, I seem to have an [issue](https://github.com/ros-planning/navigation2_tutorials/issues/25) with my code.
It would be very helpful thanks.

philison, 2021-04-28:
Hi @root-robot, I just pushed the tutorial code that I used with _ROS2 Foxy_ into a [repo](https://github.com/philison/navigation2_tutorials_foxy_temp). Let me know if that helps you. I haven't used the code in a long time, so I am not 100% sure that everything works.

root-robot, 2021-04-28:
Hi @philison, thanks for the quick reply.

Unfortunately, the error is the same as mentioned above [link](https://github.com/ros-planning/navigation2_tutorials/issues/25).
I will try to fix the error, but if you have any clue on how to solve it kindly revert back.
Also if you have written/know any other sample planner plugin kindly provide some references.

Regards 
@root-robot