ros-navigation/navigation2_tutorials #36: Planner plugin - build

aghari19, 2021-09-29:
Hi, 
I am trying to run the planner plugging in ROS2 Foxy. I am trying to run the straightline_planner [example](https://navigation.ros.org/plugin_tutorials/docs/writing_new_nav2planner_plugin.html) in a workspace but not able to proceed past these two errors. 
This is the github repo I am using: [github](https://github.com/ros-planning/navigation2_tutorials)
[image]

Can someone help with these please?

SteveMacenski, 2021-09-29:
These tutorials are built for galactic and newer.

closed 2021-09-29 by SteveMacenski

aghari19, 2021-10-06:
I see. Thank you for letting me know about it. 

I have a followup question. I downloaded ROS2 Galactic since the examples are for Galactic and up. After downloading ROS2 Galactic I tried to install and build Nav2 for Galactic. 

When I build Nav2 for source I get the following error on 24/34. 
I did the following things:
I followed the tutorial [here](https://navigation.ros.org/build_instructions/index.html):

When I had to execute the following commands:
[image]

I saw on github there was not a galactic-devel branch to clone from. Therefore, I cloned the galactic branch and went ahead with the build. While building the following error was encountered:
[image]

Can someone please help me with this issue or did I clone the wrong branch altogether. I have looked for help online by googling but nothing has worked. 

I truly appreciate the help, 
Thanks,