ros-navigation/navigation2_tutorials #39: Pointcloud2 rotated in Rviz

ViniciusAbrao, 2021-11-04:
I am follow the tutorials in the nav2:  https://navigation.ros.org/setup_guides/sensors/setup_sensors.html    ... and I am finding the pointcloud2 rotated in the rviz.
I double check copying the urdf from the link: https://github.com/ros-planning/navigation2_tutorials/blob/master/sam_bot_description/src/description/sam_bot_description.urdf  and it really has shown the rotated pointcloud2 in the rviz.

ViniciusAbrao, 2021-11-05:
Looks like a problem in the gazebo_ros_pkgs. 
I already open a new issue here:
https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1329

SteveMacenski, 2021-11-08:
Did you use the code exactly as shown in the repo that @jd-deleon has merged? That's the same code used to generate these images, so that makes me think potentially something's been modified on your end or using different distributions where different patches have been made.  @jd-deleon what distro did you use to make these demos? Can you confirm that if you use what's in the repo right now, you get the correct result on your end?

ViniciusAbrao, 2021-11-09:
Hi @SteveMacenski ... I cloned the master branch of the navigation2_tutorials as it is now... Also I am using Foxy, but I believe that gazebo_ros_camera plugin is similar in both Galactic and Foxy, so that makes me think about that the source of the problem is at gazebo_ros_pkgs's plugin. Besides that, the example is very simple, it just spawn the robot in Gazebo with the depth camera, and it has not any relation with the nav2 at this point of tutorial.

SteveMacenski, 2021-11-09:
I think these were developed on Galactic, but will need to wait to find out from @jd-deleon. It may have been a bug fixed in later distros for all I'm aware. 

Should we close this at this point or do you want to  keep this open?

ViniciusAbrao, 2021-11-09:
Please, let's wait the @jd-deleon answer. If it is a plugin error at gazebo_ros_pkgs we close here and I keep it only here: ros-simulation/gazebo_ros_pkgs#1329 ... in the meanwhile I am going to double check it at Galactic.

ViniciusAbrao, 2021-11-09:
I can see now that in order to generate the pointcloud2 image in the tutorial, the fixed frame is changed to camera_depth_frame:

https://navigation.ros.org/_images/demo_pointcloud2_rviz.png

Obviously, by changing the reference frame the answer is correct. Otherwise, I just checked that also in Galactic, the image is rotated.

[image]

Looks like it is not a bug fixed in a later distro @SteveMacenski. So, If you and @jd-deleon agree that the bug is really at the gazebo_ros_pkgs's plugin, feel free to close it here. Otherwise, please let me know how to fix it. Tks.

SteveMacenski, 2021-11-09:
Lets see what they say on the ticket you filed, but if you don't get a response or there won't be a change, then a PR here to fix the rotation in the URDF would be appreciated

jd-deleon, 2021-11-09:
Just wanted to confirm that the tutorial and images were taken using Galactic version. We did notice this behavior when making the tutorial but we didn't dive deep to why it was happening. In the end we decided to just change reference frames since it was just used for validating that the simulation was somewhat working. Good catch on identifying the root cause, I think we can forward the issue to the plugin team.

SteveMacenski, 2021-11-09:
Well sounds like a long standing issue we should probably fix the configuration for. @jd-deleon I think all you need to do is mess with the Roll Pitch Yaw for the mounting angle of the sensor so that things are correct in the global frame.

ViniciusAbrao, 2021-11-09:
Hi @SteveMacenski ... I have implemented and tested a solution here:

https://github.com/ViniciusAbrao/ros2_depth_cam_plugin

Feel free to close it here, if you want, since it is not a nav2 issue. We may continue to addressee in the issue:

https://github.com/ros-simulation/gazebo_ros_pkgs/issues/1329

SteveMacenski, 2021-11-09:
I don't think that the issue here is probably gazebo plugins, I think its our configuration. Cameras/CV have different frame conventions than robotics, so that frame choice makes sense. Its a little... unexpected, but reasonable. I don't think ultimately it will be merged because they'll want to keep their frame choices. So I think the best answer is actually just to fix this URDF to conform to that standard

ViniciusAbrao, 2021-11-09:
I could see that in ROS1, the preferred plugin was libgazebo_ros_openni_kinect.so
In that case, the urdf had been simple adapted, as following:

  <joint name="camera_optical_joint" type="fixed">
    <origin xyz="0 0 0" rpy="-1.5707 0 -1.5707"/>
    <parent link="camera"/>
    <child link="camera_link_optical"/>
  </joint>

We may find this instruction at Gazebo Tutorials. 

But now, in ROS2, we do not have libgazebo_ros_openni_kinect.so, and in this case, the camera plugin should work similarly. However, I am not sure if it is my fault, I did not find how to adapt the URDF, even including the optical frame and setting it in:

<frameName>camera_depth_frame</frameName>

SteveMacenski, 2021-11-09:
https://github.com/ros-planning/navigation2_tutorials/blob/master/sam_bot_description/src/description/sam_bot_description.urdf#L314-L348

The `origin` is what you're wanting to play with, in the RPY fields

ViniciusAbrao, 2021-11-09:
Exact... also I have tested with -pi/2 and -pi/2 .

I could see that in ROS1, the preferred plugin was libgazebo_ros_openni_kinect.so
In that case, the urdf had been simple adapted, as following:

joint name="camera_optical_joint" type="fixed"
origin xyz="0 0 0" rpy="-1.5707 0 -1.5707"
parent link="camera"
child link="camera_link_optical"

We may find this instruction at Gazebo Tutorials.

But now, in ROS2, we do not have libgazebo_ros_openni_kinect.so, and in this case, the camera plugin should work similarly. However, I am not sure if it is my fault, I did not find how to adapt the URDF, even including the optical frame and setting it in:

frameName>camera_depth_frame

SteveMacenski, 2021-11-09:
I dont understand what you mean / the issue

SteveMacenski, 2021-11-09:
Did you recompile?

ViniciusAbrao, 2021-11-09:
For sure!!

ViniciusAbrao, 2021-11-09:
Lets see if they have some different point of view. If it is not necessary the PR, probably they know how to edit the URDF in order to correct it.

arubenecia, 2021-11-22:
Please try the following changes in the URDF (applied to the linked PR):

1. Set rpy of `camera_depth_joint` to `"${-pi/2} 0 ${-pi/2}"` (L345). This should rotate the `camera_depth_frame` TF and have the following coordinate:  x - right, y - down, z – forward. 
2. Change `<frameName>camera_depth_frame</frameName>` to `<frame_name>camera_depth_frame</frame_name>` (L370). This should publish the pointcloud to camera_depth_frame. 

The pointcloud should now be shown in front of sam_bot in RViz:

[image]

ViniciusAbrao, 2021-11-22:
Brilliant ... The value of rpy angles I had tried already, but the secret is at "frame_name" instead of "frameName" ... I can not believe on that !!! Congrats and tks.

PR #40 (2021-11-23): Fix sam_bot_description.urdf to visualize pointcloud along x-axis in RViz
Edit coordinate frame of camera_depth_frame to x - right, y - down, z – forward. 
Fix frame name tag in gazebo camera plugin

Fixes #39
closed 2021-11-22 by ViniciusAbrao