ros-navigation/navigation2_tutorials #42: nav2_pure_pursuit_controller pkg not in master branch

figoowen2003, 2021-12-03:
I cloned the whole repo to my local directory, but I could not find nav2_pure_pursuit_controller pkg in it. I check this pkg on website and find that it could not show in master branch, instead it will appear in a branch named 126902457c5c646b136569886d6325f070c1073d. Will you please deal with it, please? 
[image]
[image]

SteveMacenski, 2021-12-03:
That is deprecated. It's been further developed on and merged into Nav2 proper as a new controller plugin for use https://github.com/ros-planning/navigation2/tree/main/nav2_regulated_pure_pursuit_controller

See our plug documentation and migration guides that cover this extensively https://navigation.ros.org/

SteveMacenski, 2021-12-03:
I also glanced at your website (figoowen2003.github.io) and it contains a **great deal** of copyright infringement. Please remove blog posts that clearly copy + paste content from our documentation without attribution (or add proper attribution).

closed 2021-12-03 by SteveMacenski