ros-navigation/navigation2_tutorials #51: ‘using SharedPtr = class std::shared_ptr<rclcpp_lifecycle::LifecycleNode>’ {aka ‘class std::shared_ptr<rclcpp_lifecycle::LifecycleNode>’} has no member named ‘lock’

ct2034, 2022-05-09:
When try to build the repo as is under `foxy` using `colcon build`, I get:

```
Starting >>> nav2_costmap_filters_demo
Starting >>> nav2_gradient_costmap_plugin
Starting >>> nav2_sms_recovery
Starting >>> nav2_straightline_planner
Starting >>> sam_bot_description
Finished <<< nav2_costmap_filters_demo [0.91s]                                  
Finished <<< sam_bot_description [1.14s]                                         
--- stderr: nav2_gradient_costmap_plugin                                             
/home/chenkel/ros2/nav2_ws/src/navigation2_tutorials/nav2_gradient_costmap_plugin/src/gradient_layer.cpp: In member function ‘virtual void nav2_gradient_costmap_plugin::GradientLayer::onInitialize()’:
/home/chenkel/ros2/nav2_ws/src/navigation2_tutorials/nav2_gradient_costmap_plugin/src/gradient_layer.cpp:70:21: error: ‘using SharedPtr = class std::shared_ptr<rclcpp_lifecycle::LifecycleNode>’ {aka ‘class std::shared_ptr<rclcpp_lifecycle::LifecycleNode>’} has no member named ‘lock’
   70 |   auto node = node_.lock();
      |                     ^~~~
make[2]: *** [CMakeFiles/nav2_gradient_costmap_plugin_core.dir/build.make:63: CMakeFiles/nav2_gradient_costmap_plugin_core.dir/src/gradient_layer.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/nav2_gradient_costmap_plugin_core.dir/all] Error 2
make: *** [Makefile:141: all] Error 2
---
Failed   <<< nav2_gradient_costmap_plugin [6.27s, exited with code 2]
Aborted  <<< nav2_straightline_planner [10.4s]                                         
Aborted  <<< nav2_sms_recovery [11.1s]                                 

Summary: 2 packages finished [11.3s]
  1 package failed: nav2_gradient_costmap_plugin
  2 packages aborted: nav2_sms_recovery nav2_straightline_planner
  3 packages had stderr output: nav2_gradient_costmap_plugin nav2_sms_recovery nav2_straightline_planner
```

SteveMacenski, 2022-05-09:
These are not for foxy, they're for Galactic and newer, sorry abotu that. You can always change the APIs to use the old shared pointers  versus weak pointers.

closed 2022-05-09 by SteveMacenski

ct2034, 2022-05-09:
Don't you think this should be mentioned somewhere? I guessed that already. But if I look at the repo, it only had a main branch, and no other hint on the targeted distro. And also the nav2 docs simply tells you to install the distro you want (at https://navigation.ros.org/build_instructions/index.html#install)

SteveMacenski, 2022-05-09:
Those kinds of changes won't be made in the future so its just temporary growing pains for a project with active development and Foxy. Sorry for the inconvenience