ros-navigation/navigation2_tutorials #56: Sam bot description tutorial - not really working

omerts, 2022-06-29:
Hi,

I have followed the first robot set up tutorial strictly, and my robot is just drifiting in different directions. After trying to move the sam robot in this repo, I get the same behavior. 
[image]

There are also some misconfiguration:
1) Because of Gazebo's diff driver plugin's namespace, no cmd_vel commands are received
2) bt_navigator & controller_server are registered to /odom, while gazebo is publishing to /demo/odom
3) There is an issue with the time on the tf frames because not all nodes are using the sim time parameter. 

For the misconfigurations I have created a PR:

For the reason the robot is flying away, I have been batteling it, being able to only mitigate it, but not solve it.

elgarbe, 2024-02-03:
I've tried using the nav2_params.yaml provided with sam_bot (with [this PR added](https://github.com/ros-planning/navigation2_tutorials/pull/57)):

`ros2 launch nav2_bringup navigation_launch.py params_file:=xxxxx/nav2tutorial_ws/src/navigation2_tutorials/sam_bot_description/config/nav2_params.yaml`

but get this errors:

```
[bt_navigator-5] [ERROR] [1706987053.543650228] [bt_navigator]: Exception when loading BT: Error at line 14: -> Node not recognized: WouldAPlannerRecoveryHelp
[bt_navigator-5] [ERROR] [1706987053.543975397] [bt_navigator]: Error loading XML file: /opt/ros/iron/share/nav2_bt_navigator/behavior_trees/navigate_to_pose_w_replanning_and_recovery.xml 
[lifecycle_manager-8] [ERROR] [1706987053.545413334] [lifecycle_manager_navigation]: Failed to change state for node: bt_navigator
[lifecycle_manager-8] [ERROR] [1706987053.545639442] [lifecycle_manager_navigation]: Failed to bring up all requested nodes. Aborting bringup.
```

elgarbe, 2024-02-03:
adding:

```
      - nav2_are_error_codes_active_condition_bt_node
      - nav2_would_a_controller_recovery_help_condition_bt_node
      - nav2_would_a_planner_recovery_help_condition_bt_node
      - nav2_would_a_smoother_recovery_help_condition_bt_node   
```
to nav2_params.yaml on sam_bot_description solve previos problem.
Now I can see costmaps as expected by tutorial.

omerts, 2024-08-12:
@PrakritTyagi Unfortunately not

SteveMacenski, 2024-11-18:
PR merged; branches created for humble separately with those parts of the config file removed.

closed 2024-11-18 by SteveMacenski