ros-navigation/navigation2_tutorials #58: bt_navigator error on tutorial file

andrejpadilha2, 2022-07-01:
There was probably an update on nav2 that wasn't updated in the basic setup here: https://navigation.ros.org/setup_guides/footprint/setup_footprint.html

The configuration file `nav2_params.yaml` needs one line to work, under the `bt_navigator`, approx. line 50:
just add `default_bt_xml_filename: "navigate_w_replanning_and_recovery.xml"` after `groot_zmq_server_port: 1667` and before `plugin_lib_names:`.

I compared with the original file that was installed with the nav2 package.

SteveMacenski, 2022-07-01:
Can you submit a PR? Actually the ZMQ ports are no longer relevant, that feature was deprecated.

andrejpadilha2, 2022-07-01:
Sure, just submitted it. Hope it helps!

closed 2023-10-06 by SteveMacenski

antoszy, 2024-04-12:
@SteveMacenski Is real-time BT graph visualization still possible with ZMQ ports depreciated?