ros-navigation/navigation2_tutorials #60: Keepout zone tutorial not working

Salvo-Dippolito, 2022-07-04:
After following all the instructions in https://navigation.ros.org/tutorials/docs/navigation2_with_keepout_filter.html I can't get gazebo to load the world and the  robot, only the sensor raytrace is shown and no errors seem to pop up in the terminal.
I tried following the steps explained in this tutorial to apply them on a project that I'm doing on webots but the keepout_mask is not loading on rviz. Everything seems to be fine  as in no obvious errors pop up in the terminal but the costmap filter won't show up no matter what I try. One weird thing I noticed is that after changing the turtlebot3_world.yaml filename in the file nav2_params nothing changes, no errors show up, maybe the whole file is not being used properly?

Salvo-Dippolito, 2022-07-04:
Update: after taking all project files under the same workspave I am now reading this error message:

recoveries_server-16] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-19] [INFO] [1656977833.538670409] [lifecycle_manager_navigation]: Creating
[lifecycle_manager-19] [INFO] [1656977833.540779978] [lifecycle_manager_navigation]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [1656977833.421280931] [lifecycle_manager_costmap_filters]: Creating
[lifecycle_manager-1] [INFO] [1656977833.423460407] [lifecycle_manager_costmap_filters]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [1656977833.432177317] [lifecycle_manager_costmap_filters]: Starting managed nodes bringup...
[lifecycle_manager-1] [INFO] [1656977833.432243155] [lifecycle_manager_costmap_filters]: Configuring filter_mask_server
[lifecycle_manager-1] [ERROR] [1656977833.462656108] [lifecycle_manager_costmap_filters]: Failed to change state for node: filter_mask_server
[lifecycle_manager-1] [ERROR] [1656977833.462721179] [lifecycle_manager_costmap_filters]: Failed to bring up all requested nodes. Aborting bringup.
[map_server-2] [INFO] [1656977833.434126122] [filter_mask_server]: 
[map_server-2] 	filter_mask_server lifecycle node launched. 
[map_server-2] 	Waiting on external lifecycle transitions to activate
[map_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1656977833.434276163] [filter_mask_server]: Creating
[map_server-2] [INFO] [1656977833.461476941] [filter_mask_server]: Configuring
[map_server-2] [ERROR] [1656977833.461709952] []: Caught exception in callback for transition 10
[map_server-2] [ERROR] [1656977833.461782828] []: Original error: parameter 'yaml_filename' is not initialized
[map_server-2] [WARN] [1656977833.461826341] []: Error occurred while doing error handling.
[map_server-2] [FATAL] [1656977833.461850148] [filter_mask_server]: Lifecycle node filter_mask_server does not have error state implemented

SteveMacenski, 2022-07-05:
>  can't get gazebo to load the world and the robot

That doesn't really have anything to do with this feature. This does not modify any simulation related files so there is something wrong with your bringup.

closed 2022-07-05 by SteveMacenski

SteveMacenski, 2022-07-06:
> The only issue now is that the robot finds a very close path to the keepout zones and ends up getting stuck on their borders

You could place it before the inflation layer if you'd like to inflate the keepout zone borders so that it incentivizes keeping some distance. 

Further, you could also set in your keepout zone file a lighter color (e.g. lower cost) area around the critical keepout zone to punish going close to the border of the keepout zone, but still make it technically possible if absolutely required.

Salvo-Dippolito, 2022-07-19:
Thank you very much, greying out the borders of the keepout zones worked out perfectly in the end.