ros-navigation/navigation2_tutorials #62: Error at colcon build in making costmap 2d plugin

Kushagra-05, 2022-08-24:
[image]
I am getting 3 errors, firstly, it says that "gradient_layer.hpp: No such file" whereas it exists in the mentioned location.
The other two errors, are CMake ones, please look at the screenshot attached.

Kushagra-05, 2022-08-27:
The issue came because i was a little confused with two folders with same name, nav2_gradient_costmap_plugin folder has an include folder which again has a nav2_gradient_costmap_plugin folder which has to be the place for gardient_layer.hpp file, exactly the same as github repo shows.
It was my digression that gave the 1st error (in red)

closed 2022-08-27 by Kushagra-05

Namo98, 2023-08-23:
> The issue came because i was a little confused with two folders with same name, nav2_gradient_costmap_plugin folder has an include folder which again has a nav2_gradient_costmap_plugin folder which [...]

Hey Kushagra. Did the nav2_gradientcostmap_pliugin works for you?

i tried to add the nav2_gradient_costmap_plugin but the robot can still move:
https://streamable.com/3g2vw8