ros-navigation/navigation2_tutorials #64: Robot_Localization_Demo: problem with rviz2 and ekf_node Warning: TF_OLD_DATA ignoring data from the past

wongearth, 2022-11-18:
I am trying to follow the sam_bot tutorial but got stuck at the robot_localization demo using ekf_node. Below is the error/warning when running
 
ros2 launch sam_bot_description display.launch.py

By running "ros2 run tf2_ros tf2_echo odom base_link",  similar errors/warning occured.  In rviz the odom franme seems to drift. I reinstalled ros2, navigation2 both binary and build from source...no luck.  Anyone know what the problem is?... Thanks

[rviz2-6] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-6]          at line 292 in /home/xxx/ros2_humble/src/ros2/geometry2/tf2/src/buffer_core.cpp
[ekf_node-5] Warning: TF_OLD_DATA ignoring data from the past for frame drivewhl_l_link at time 755.451000 according to authority Authority undetectable
[ekf_node-5] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ekf_node-5]          at line 292 in /home/xxx/ros2_humble/src/ros2/geometry2/tf2/src/buffer_core.cpp
[rviz2-6] Warning: TF_OLD_DATA ignoring data from the past for frame drivewhl_l_link at time 755.451000 according to authority Authority undetectable
[rviz2-6] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[rviz2-6]          at line 292 in /home/xxx/ros2_humble/src/ros2/geometry2/tf2/src/buffer_core.cpp
[ekf_node-5] Warning: TF_OLD_DATA ignoring data from the past for frame drivewhl_r_link at time 755.461000 according to authority Authority undetectable
[ekf_node-5] Possible reasons are listed at http://wiki.ros.org/tf/Errors%20explained
[ekf_node-5]          at line 292 in /home/xxx/ros2_humble/src/ros2/geometry2/tf2/src/buffer_core.cpp
[rviz2-6] Warning: TF_OLD_DATA ignoring data from the past for frame drivewhl_r_link at time 755.461000 according to authority Authority undetectable

Sraobot, 2022-11-18:
I am also getting this error in "Adding Gazebo Plugins to a URDF" tutorial from Setting Up Odometry.

Wellington-Noberto, 2023-03-15:
Could you check if any Node definition ends with a comma?
Especially the 'joint_state_publisher'

If you forgot the comma this could lead to an error in the joint state publisher node
`joint_state_publisher_node = Node(
        package='joint_state_publisher',
        executable='joint_state_publisher',
        name='joint_state_publisher',
        condition=UnlessCondition(LaunchConfiguration('gui')),`
)

TanJunKiat, 2023-06-18:
The issue is the the joint_state_publisher is not publishing in sim_time

I managed to resolve this issue with the following code:

```
joint_state_publisher_node = launch_ros.actions.Node(
        package='joint_state_publisher',
        executable='joint_state_publisher',
        name='joint_state_publisher',
        parameters=[{'use_sim_time': LaunchConfiguration('use_sim_time')}],
    )
```

which adds the use_sim_time parameter to the joint_state_publisher node.

Jason-Lee0, 2023-09-15:
@TanJunKiat  Sorry to bother you. I updated the command (use_sim_time) in Joint_state_publisher_node , but it still showed the same  warning log . 

Warning: TF_OLD_DATA ignoring data from the past for frame drivewhl_l_link at time 419.418000 according to authority Authority undetectable
  
Is there any other way to avoid this ?
Thanks for your help. 

Have a nice day!!

TanJunKiat, 2023-09-15:
@Jason-Lee0 Hello Jason.

Have you tried checking your SDF?

`<gazebo>
  <plugin name='diff_drive' filename='libgazebo_ros_diff_drive.so'>
    <ros>
      <namespace>/demo</namespace>
    </ros>
    
    <update_rate>60</update_rate>
    
    
    <command_topic>cmd_vel</command_topic>

    
    <left_joint>drivewhl_l_joint</left_joint>
    <right_joint>drivewhl_r_joint</right_joint>

    
    <wheel_separation>0.4</wheel_separation>
    <wheel_diameter>0.2</wheel_diameter>

    
    <max_wheel_torque>40</max_wheel_torque>
    <max_wheel_acceleration>2.0</max_wheel_acceleration>

    
    <publish_odom>true</publish_odom>
    <publish_odom_tf>true</publish_odom_tf>
    <publish_wheel_tf>true</publish_wheel_tf>

    <odometry_topic>odom</odometry_topic>
    <odometry_frame>odom</odometry_frame>
    <robot_base_frame>base_link</robot_base_frame>
  </plugin>
</gazebo>`

It may be the lack of definition of the tf for the wheels.

Charifou, 2024-02-20:
I confirm that you should add 'use_sim_time': True to all your nodes inside your launch file and its included launch files using:

parameters=[{'use_sim_time': True}, config_ekf_yaml].

Here it comes from ekf_node and rviz2.
If you have nodes that share a yaml file you can directly add 'use_sim_time' in the parameters list and use /** instead of node names.

/**:
  ros__parameters:
    use_sim_time : True

MansiP441, 2024-07-08:
> I confirm that you should add 'use_sim_time': True to all your nodes inside your launch file and its included launch files using:
> 
> parameters=[{'use_sim_time': True}, config_ekf_yaml].
> 
> Here it comes from ekf_node and rviz2. If you have nodes which share a yaml file you can directly add 'use_sim_time' in the parameters list and use /** instead of node names.
> 
> /**: ros__parameters: use_sim_time : True

I agree with @Charifou. 
Additionally, you can use rqt (dynamic reconfigure) to find out which nodes do not have use_sim_time set to True. 
In my case, setting 'use_sim_time' to True for joint_state_broadcaster node solved the problem.