ros-navigation/navigation2_tutorials #66: Activating keepout and speed filter together.

dharanish-nh, 2023-05-03:
Hello,

I'm working on a project that requires the use of two costmap filters at the same time: the Keepout filter and the Speed filter. I've been looking through the documentation and I'm having trouble figuring out how to activate both filters at once. Could someone please provide guidance on how to do this?

Thank you for your help.

I tried to make some changes to the code however it does not work.

```
def generate_launch_description():
    # Get the launch directory
    costmap_filters_demo_dir = get_package_share_directory('nav2_costmap_filters_demo')

    # Create our own temporary YAML files that include substitutions
    lifecycle_nodes = ['filter_mask_server', 'costmap_filter_info_server']

    # Parameters
    namespace = LaunchConfiguration('namespace')
    use_sim_time = LaunchConfiguration('use_sim_time')
    autostart = LaunchConfiguration('autostart')
    params_file_keepout = LaunchConfiguration('params_file_keepout')
    params_file_speed = LaunchConfiguration('params_file_speed')
    mask_yaml_file_keepout = LaunchConfiguration('mask_keepout')
    mask_yaml_file_speed = LaunchConfiguration('mask_speed')

    # Declare the launch arguments
    declare_namespace_cmd = DeclareLaunchArgument(
        'namespace',
        default_value='',
        description='Top-level namespace')

    declare_use_sim_time_cmd = DeclareLaunchArgument(
        'use_sim_time',
        default_value='true',
        description='Use simulation (Gazebo) clock if true')

    declare_autostart_cmd = DeclareLaunchArgument(
        'autostart', default_value='true',
        description='Automatically startup the nav2 stack')

    declare_params_file_keepout_cmd = DeclareLaunchArgument(
            'params_file_keepout',
            default_value=os.path.join(get_package_share_directory('nav2_costmap_filters_demo'), 'params', 'keepout_params.yaml'),
            description='Full path to the ROS2 parameters file for the Keepout filter')

    declare_params_file_speed_cmd = DeclareLaunchArgument(
            'params_file_speed',
            default_value=os.path.join(get_package_share_directory('nav2_costmap_filters_demo'), 'params', 'speed_params.yaml'),
            description='Full path to the ROS2 parameters file for the Speed filter')

    declare_mask_yaml_file_keepout_cmd = DeclareLaunchArgument(
            'mask_keepout',
            default_value=os.path.join(get_package_share_directory('nav2_costmap_filters_demo'), 'maps', 'fullmap_keepout.yaml'),
            description='Full path to filter mask yaml file to load for the Keepout filter')

    declare_mask_yaml_file_speed_cmd = DeclareLaunchArgument(
            'mask_speed',
            default_value=os.path.join(get_package_share_directory('nav2_costmap_filters_demo'), 'maps', 'speed.yaml'),
            description='Full path to filter mask yaml file to load for the Speed filter')

    # Make re-written yaml for Keepout filter
    param_substitutions_keepout = {
        'use_sim_time': use_sim_time,
        'yaml_filename': mask_yaml_file_keepout}

    configured_params_keepout = RewrittenYaml(
        source_file=params_file_keepout,
        root_key=namespace,
        param_rewrites=param_substitutions_keepout,
        convert_types=True)
[... 82 lines cut ...]
[costmap_filter_info_server-4] 	costmap_filter_info_server_2 lifecycle node launched. 
[costmap_filter_info_server-4] 	Waiting on external lifecycle transitions to activate
[costmap_filter_info_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1683156616.687254944] [filter_mask_server]: 
[map_server-2] 	filter_mask_server lifecycle node launched. 
[map_server-2] 	Waiting on external lifecycle transitions to activate
[map_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1683156616.687365271] [filter_mask_server]: Creating
[costmap_filter_info_server-3] [INFO] [1683156616.746267633] [costmap_filter_info_server_1]: 
[costmap_filter_info_server-3] 	costmap_filter_info_server_1 lifecycle node launched. 
[costmap_filter_info_server-3] 	Waiting on external lifecycle transitions to activate
[costmap_filter_info_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-1] [INFO] [1683156616.773070866] [lifecycle_manager_costmap_filters]: Creating and initializing lifecycle service clients
[lifecycle_manager-1] [INFO] [1683156618.847922028] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156620.848138293] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156622.848298710] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156624.848502387] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156626.848733649] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156628.848942898] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156630.849106236] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156632.849306747] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156634.849465813] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...
[lifecycle_manager-1] [INFO] [1683156636.849632568] [lifecycle_manager_costmap_filters]: Waiting for service costmap_filter_info_server/get_state...

```

SteveMacenski, 2023-05-04:
Simply follow the instructions for both individually, they should not conflict in any way ( @AlexeyMerzlyakov )

AlexeyMerzlyakov, 2023-05-04:
Most probably, the problem is that lifecycle manager does not know about two simultaneously running filter info servers, and thus can not change their state which seem in logs:
```
lifecycle_nodes = ['filter_mask_server', 'costmap_filter_info_server']
```

Try to change this to:
```
lifecycle_nodes = ['filter_mask_server', 'costmap_filter_info_server_1', 'costmap_filter_info_server_2']
```

dharanish-nh, 2023-05-04:
I made those changes to the code. The code works however I am not receiving the map. Again if run them individually it works.
should I make any changes to the YAML files as of now I have not made any changes.

output log of costmap_filter_info_2.launch.py
```
ros2 launch nav2_costmap_filters_demo costmap_filter_info_2.launch.py 
[INFO] [launch]: All log files can be found below /home/bot/.ros/log/2023-05-04-13-30-51-100544-bot-28556
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [lifecycle_manager-1]: process started with pid [28560]
[INFO] [map_server-2]: process started with pid [28562]
[INFO] [costmap_filter_info_server-3]: process started with pid [28564]
[INFO] [costmap_filter_info_server-4]: process started with pid [28566]
[lifecycle_manager-1] [INFO] [1683199851.312745679] [lifecycle_manager_costmap_filters]: Creating
[lifecycle_manager-1] [INFO] [1683199851.314517811] [lifecycle_manager_costmap_filters]: Creating and initializing lifecycle service clients
[costmap_filter_info_server-3] [INFO] [1683199851.317287383] [costmap_filter_info_server]: This is costmap filter info publisher
[costmap_filter_info_server-4] [INFO] [1683199851.318580692] [costmap_filter_info_server]: This is costmap filter info publisher
[map_server-2] [INFO] [1683199851.326218103] [filter_mask_server]: 
[map_server-2] 	filter_mask_server lifecycle node launched. 
[map_server-2] 	Waiting on external lifecycle transitions to activate
[map_server-2] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[map_server-2] [INFO] [1683199851.326313268] [filter_mask_server]: Creating
[costmap_filter_info_server-3] [INFO] [1683199851.327711099] [costmap_filter_info_server_1]: 
[costmap_filter_info_server-3] 	costmap_filter_info_server_1 lifecycle node launched. 
[costmap_filter_info_server-3] 	Waiting on external lifecycle transitions to activate
[costmap_filter_info_server-3] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[costmap_filter_info_server-4] [INFO] [1683199851.330852765] [costmap_filter_info_server_2]: 
[costmap_filter_info_server-4] 	costmap_filter_info_server_2 lifecycle node launched. 
[costmap_filter_info_server-4] 	Waiting on external lifecycle transitions to activate
[costmap_filter_info_server-4] 	See https://design.ros2.org/articles/node_lifecycle.html for more information.
[lifecycle_manager-1] [INFO] [1683199851.519095705] [lifecycle_manager_costmap_filters]: Starting managed nodes bringup...
[lifecycle_manager-1] [INFO] [1683199851.519176336] [lifecycle_manager_costmap_filters]: Configuring filter_mask_server
[map_server-2] [INFO] [1683199851.519815644] [filter_mask_server]: Configuring
[map_server-2] [INFO] [map_io]: Loading yaml file: /home/bot/tutorials_ws/install/nav2_costmap_filters_demo/share/nav2_costmap_filters_demo/maps/fullmap_keepout.yaml
[map_server-2] [DEBUG] [map_io]: resolution: 0.05
[map_server-2] [DEBUG] [map_io]: origin[0]: -26.8
[map_server-2] [DEBUG] [map_io]: origin[1]: -3.56
[map_server-2] [DEBUG] [map_io]: origin[2]: 0
[map_server-2] [DEBUG] [map_io]: free_thresh: 0.196
[map_server-2] [DEBUG] [map_io]: occupied_thresh: 0.65
[map_server-2] [DEBUG] [map_io]: mode: trinary
[map_server-2] [DEBUG] [map_io]: negate: 0
[map_server-2] [INFO] [map_io]: Loading image_file: /home/bot/tutorials_ws/install/nav2_costmap_filters_demo/share/nav2_costmap_filters_demo/maps/restricted_zone.pgm
[map_server-2] [DEBUG] [map_io]: Read map /home/bot/tutorials_ws/install/nav2_costmap_filters_demo/share/nav2_costmap_filters_demo/maps/restricted_zone.pgm: 1207 X 591 map @ 0.05 m/cell
[lifecycle_manager-1] [INFO] [1683199851.652817587] [lifecycle_manager_costmap_filters]: Configuring costmap_filter_info_server_1
[costmap_filter_info_server-3] [INFO] [1683199851.653118615] [costmap_filter_info_server_1]: Configuring
[lifecycle_manager-1] [INFO] [1683199851.655315007] [lifecycle_manager_costmap_filters]: Configuring costmap_filter_info_server_2
[costmap_filter_info_server-4] [INFO] [1683199851.655602376] [costmap_filter_info_server_2]: Configuring
[lifecycle_manager-1] [INFO] [1683199851.658408431] [lifecycle_manager_costmap_filters]: Activating filter_mask_server
[map_server-2] [INFO] [1683199851.658632930] [filter_mask_server]: Activating
[map_server-2] [INFO] [1683199851.659224932] [filter_mask_server]: Creating bond (filter_mask_server) to lifecycle manager.
[lifecycle_manager-1] [INFO] [1683199851.762686287] [lifecycle_manager_costmap_filters]: Server filter_mask_server connected with bond.
[lifecycle_manager-1] [INFO] [1683199851.762727316] [lifecycle_manager_costmap_filters]: Activating costmap_filter_info_server_1
[costmap_filter_info_server-3] [INFO] [1683199851.763032972] [costmap_filter_info_server_1]: Activating
[costmap_filter_info_server-3] [INFO] [1683199851.763110993] [costmap_filter_info_server_1]: Creating bond (costmap_filter_info_server_1) to lifecycle manager.
[lifecycle_manager-1] [INFO] [1683199851.867724324] [lifecycle_manager_costmap_filters]: Server costmap_filter_info_server_1 connected with bond.
[lifecycle_manager-1] [INFO] [1683199851.867771251] [lifecycle_manager_costmap_filters]: Activating costmap_filter_info_server_2
[costmap_filter_info_server-4] [INFO] [1683199851.868342757] [costmap_filter_info_server_2]: Activating
[costmap_filter_info_server-4] [INFO] [1683199851.868430797] [costmap_filter_info_server_2]: Creating bond (costmap_filter_info_server_2) to lifecycle manager.
[... cut ...]
[planner_server-20] [WARN] [1683199863.513337844] [global_costmap.global_costmap]: SpeedFilter: Filter mask was not received
[controller_server-18] [WARN] [1683199864.861485278] [local_costmap.local_costmap]: KeepoutFilter: Filter mask was not received
[planner_server-20] [WARN] [1683199866.513113897] [global_costmap.global_costmap]: KeepoutFilter: Filter mask was not received
[planner_server-20] [WARN] [1683199866.513147583] [global_costmap.global_costmap]: SpeedFilter: Filter mask was not received
```

AlexeyMerzlyakov, 2023-05-05:
@dharanish-nh, you have underconfigured system, therefore filters are not operating well. In order to make speed and keepout filters are working altogether, please make sure that:
1. Along with two costmap filters info server in the `nav2_costmap_filters_demo/launch/costmap_filter_info.launch.py` there are two different map servers are also set (which I did not seen in the launch-file from above). The rule is simple: one costmap filter info and one map server per each cosmap filter. Please also make sure, that their names are also different, to keep namespaces to be consistent.
2. You have correctly configured both `params/speed_params.yaml` and `params/keepout_params.yaml` params-files setting proper namespaces for each costmap filter info and map server.
3. In `nav2_bringup/params/nav2_params.yaml` both filters were set and configured correctly each one for its own mask topic:
```
diff --git a/nav2_bringup/params/nav2_params.yaml b/nav2_bringup/params/nav2_params.yaml
index 019bbb39..bf32a249 100644
--- a/nav2_bringup/params/nav2_params.yaml
+++ b/nav2_bringup/params/nav2_params.yaml
@@ -116,6 +116,7 @@ controller_server:
     min_y_velocity_threshold: 0.5
     min_theta_velocity_threshold: 0.001
     failure_tolerance: 0.3
+    speed_limit_topic: "/speed_limit"
     progress_checker_plugin: "progress_checker"
     goal_checker_plugins: ["general_goal_checker"] # "precise_goal_checker"
     controller_plugins: ["FollowPath"]
@@ -232,6 +233,7 @@ global_costmap:
       resolution: 0.05
       track_unknown_space: true
       plugins: ["static_layer", "obstacle_layer", "inflation_layer"]
+      filters: ["keepout_filter", "speed_filter"]
       obstacle_layer:
         plugin: "nav2_costmap_2d::ObstacleLayer"
         enabled: True
@@ -253,6 +255,15 @@ global_costmap:
         plugin: "nav2_costmap_2d::InflationLayer"
         cost_scaling_factor: 3.0
         inflation_radius: 0.55
+      keepout_filter:
+        plugin: "nav2_costmap_2d::KeepoutFilter"
+        enabled: True
+        filter_info_topic: "/keepout_filter_info"
+      speed_filter:
+        plugin: "nav2_costmap_2d::SpeedFilter"
+        enabled: True
+        filter_info_topic: "/speed_filter_info"
+        speed_limit_topic: "/speed_limit"
       always_send_full_costmap: True
 
 # The yaml_filename does not need to be specified since it going to be set by defaults in launch.
```

For the reference, attaching zipped demonstration sources that work fine for me:
[nav2_costmap_filters_demo.zip](https://github.com/ros-planning/navigation2_tutorials/files/11403698/nav2_costmap_filters_demo.zip)

I've enabled keepout and speed filter simultaneously for `global_costmap` and they did work altogether as follows from the picture below:
[image]

dharanish-nh, 2023-05-05:
Thank you for all the help. It works for me as well. But I have observed one behavior when implementing keepout filter. While planning its path, if the path generated is close to the keepout zone, there is  higher chance robot base  gets stuck in the zone and causes the robot to stop navigation. How can i improve to avoid this situation?

[image]

closed 2023-05-05 by dharanish-nh