ros-navigation/navigation2_tutorials #74: interactive_waypoint_follower can not get robot_localization state

Cristian-wp, 2023-11-28:
Hello, I am trying to run the [Interactive GPS Waypoint Follower](https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html#interactive-gps-waypoint-follower).
I have launch these instruction:
**TERMINAL 1:**
`ros2 launch nav2_gps_waypoint_follower_demo gps_waypoint_follower.launch.py use_rviz:=False`
**TERMINAL 2:**
`ros2 launch nav2_gps_waypoint_follower_demo mapviz.launch.py`
**TERMINAL 3:**
`ros2 run nav2_gps_waypoint_follower_demo interactive_waypoint_follower`

When I run interactive_waypoint_follower node and click on mapviz to send waypoint I get fle ollowing output from its terminal(TERMINAL 3):
```
mapviz-1] [WARN] [1701172112.460493075] [swri_transform_util::Transformer]: Wgs84Transformer not initialized
[mapviz-1] [WARN] [1701172112.460534350] [swri_transform_util::Transformer]: Wgs84Transformer not initialized
[initialize_origin.py-2] [INFO] [1701172112.461930965] [initialize_origin]: Got NavSat message.
[initialize_origin.py-2] [INFO] [1701172112.462575108] [initialize_origin]: Origin from 'navsat' source set to 38.1614564605658, -122.45460714408775, 488.3200360890478
[initialize_origin.py-2] [INFO] [1701172112.463337905] [initialize_origin]: Successfully set origin; unsubscribing.
[mapviz-1] [INFO] [1701172112.480655151] [mapviz]: OK
[mapviz-1] [INFO] [1701172112.482309682] [mapviz]: OK
[mapviz-1] [INFO] [1701172138.485441870] [mapviz]: Point in wgs84: -122.454,38.1612
```
And this from nav2_gps_waypoint_follower_demo mapviz terminal(TERMINAL 2):
```
mapviz-1] [WARN] [1701172112.460493075] [swri_transform_util::Transformer]: Wgs84Transformer not initialized
[mapviz-1] [WARN] [1701172112.460534350] [swri_transform_util::Transformer]: Wgs84Transformer not initialized
[initialize_origin.py-2] [INFO] [1701172112.461930965] [initialize_origin]: Got NavSat message.
[initialize_origin.py-2] [INFO] [1701172112.462575108] [initialize_origin]: Origin from 'navsat' source set to 38.1614564605658, -122.45460714408775, 488.3200360890478
[initialize_origin.py-2] [INFO] [1701172112.463337905] [initialize_origin]: Successfully set origin; unsubscribing.
[mapviz-1] [INFO] [1701172112.480655151] [mapviz]: OK
[mapviz-1] [INFO] [1701172112.482309682] [mapviz]: OK
[mapviz-1] [INFO] [1701172138.485441870] [mapviz]: Point in wgs84: -122.454,38.1612
```
Why the robot_localization/get_state service is not avaiable? I have not change anything in the example code.

SteveMacenski, 2023-11-28:
https://robotics.stackexchange.com/questions/105711/nav2-interactive-waypoint-follower-example-can-not-get-robot-localization-state responded on robotics stack exchange 

**do not cross post**

closed 2023-11-28 by SteveMacenski

Cristian-wp, 2023-11-29:
@SteveMacenski I perfectly understan your point of view, and I am sorry if I am annoying with this behavior.

ZaBy10, 2023-12-19:
Hii @Cristian-wp , Did you get the solution of this ?? I am getting the same issue!!

Cristian-wp, 2024-01-08:
Hi @ZaBy10 the solution is simple...The waypoint follower is not implemented inside Huble. If you need to use that, you must switch to Iron branch

ZaBy10, 2024-01-19:
Hi @Cristian-wp , Thanks for the reply , Can i just run the package in a dockerfile consisting of ros2 iron??

Cristian-wp, 2024-01-19:
@ZaBy10 I have not do it, I have test it in a fresh Ubuntu 22 with ROS Iron.
I have not experience with docker, but if it works let me know please :)