ros-navigation/navigation2_tutorials #75: nav2_gps_waypoint_follower_demo does not work on ros2 humble!!!

ZaBy10, 2023-12-01:
Is there any way to make nav2_gps_waypoint_follower_demo package work on ros2 humble????

                                              OR

Is there any way to Navigate using gps localisation on ros2 humble or ros2 foxxy????

SteveMacenski, 2023-12-01:
This ticket does not contain any amount of necessary detail to be actionable, thus closing. These tutorials target Rolling but is likely to be useful in Humble.

closed 2023-12-01 by SteveMacenski

ZaBy10, 2023-12-04:
Hii @SteveMacenski , Could you please tell me what minor modifications must be made for it to be used in ros2 humble...
Please provide us the solution!!!

peakyquest, 2025-07-13:
I have implemented the demo in humble and I have created a PR, can you please look at it.