ros-navigation/navigation2_tutorials #78: rviz2 not receiving map (tutorial example)

PVBrodersen, 2023-12-12:
I have been following along with [this](https://navigation.ros.org/setup_guides/sensors/setup_sensors.html) tutorial, and gotten to the section regarding visualization of costmaps. However, at this point, I run into 2 issues: 

1) **Rviz2 not receiving costmaps**
[image]
As the section suggests, rviz2 is unable to receive any maps from the respective costmaps topics, even though I can confirm that data is published using topic echo: 

"
ros2 topic echo /global_costmap/costmap
header:
  stamp:
    sec: 1702383893
    nanosec: 146251016
  frame_id: map
info:
  map_load_time:
    sec: 0
    nanosec: 0
  resolution: 0.05000000074505806
  width: 44
  height: 36
  origin:
    position:
      x: -1.1546438506086725
      y: -0.6746192693179669
      z: 0.0
    orientation:
      x: 0.0
      y: 0.0
      z: 0.0
      w: 1.0
"

2) Filter dropping messages

After running "display.launch.py" it opens rviz2 and gazebo as expected, but after sometime the terminal starts outputting the following: 

> [rviz2-5] [INFO] [1702383769.153491354] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383754.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383770.177533838] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383755.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383772.160809140] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383757.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383774.177586825] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383758.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383775.168952119] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383759.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383776.161561609] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383761.146 for reason 'discarding message because the queue is full'
> [rviz2-5] [INFO] [1702383777.153654436] [rviz2]: Message Filter dropping message: frame 'map' at time 1702383763.146 for reason 'discarding message because the queue is full'

Which is also represented by a drifting map frame in rviz i.e. the frame starts moving up the z-axis. 

Any suggestions as to what this might be caused by?

PVBrodersen, 2023-12-15:
To add to this post: I got to the point where I decided to simply clone the repo, but still run into the same issues.

YinpeiDai, 2024-01-29:
Any update on this?

arad2456, 2024-01-30:
I am also having this problem. I use humble

jmk9, 2024-01-31:
I also can't solve this problem.

PVBrodersen, 2024-02-02:
> @PVBrodersen hi!
> 
> I'm facing the same problem too. which ros2 distro are you using?

I am running ROS2 Humble on Ubuntu 22.04 LTS

PVBrodersen, 2024-02-02:
However, after reading into this issue a bit more, I suspect that it may be due to the fact that the default parameter of navigation.launch.py do NOT contain the parameters specified for nav2_costmap_2d as otherwise stated in the guide. I will keep you posted if modifying this changes anything.

PVBrodersen, 2024-02-02:
> However, after reading into this issue a bit more, I suspect that it may be due to the fact that the default parameter of navigation.launch.py do NOT contain the parameters specified for nav2_costmap_2d as otherwise stated in the guide. I will keep you posted if modifying this changes anything.

Sadly a reconfiguration did not seem to fix the issue

PVBrodersen, 2024-02-13:
OK, after looking around, I have found that this issue is the same as described in another thread on the github: https://github.com/ros-planning/navigation2_tutorials/issues/64 

Essentially, the joint state publisher needs to be setup with the sim_time parameter.

After doing this, and launching everything from the same launch file I have gotten my solution to work.

dark-w, 2024-06-03:
> OK, after looking around, I have found that this issue is the same as described in another thread on the github: #64
>
> Essentially, the joint state publisher needs to be setup with the sim_time [...]

@PVBrodersen Hello, I still facing this problem...but I can't fix it even though I tried your suggestion about 'sim_time'. Can you show me your full code repo about this tutorial? Thanks a lot.

dark-w, 2024-06-03:
> @PVBrodersen Hello, I still facing this problem...but I can't fix it even though I tried your suggestion about 'sim_time'. Can you show me your full code repo about this tutorial? Thanks a lot.

Ok... Now I already been success! The main reason is also about 'use_sim_time', but I modified use_sim_time's value of global_costmap by run command 'rqt'. You just need to click a √, then all thing is right!

[image]

dark-w, 2024-06-04:
> Ok... Now I already been success! The main reason is also about 'use_sim_time', but I modified use_sim_time's value of global_costmap by run command 'rqt'. You just need to click a √, then all thing [...]

and then I find a strange thing that is the default value of 'use_sim_time' is False in the launch file 'navigation_launch.py'... So all things don't work properly even though you see the param file 'nav2_params.yaml' is right. Anyway, I fixed this last night. You can use this command:
```bash
ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
```
Okay, everything will be fine!

Makarand-Eric, 2024-07-12:
can anyone share there updated package/repo which is working?

Kavinaya7, 2024-07-29:
Hello, 
Im facing the "no map received" on rviz .. I am using ROS1 noetic and Gazebo 11
There is no data being published in the /map topic however this topic exists
What should I do?