ros-navigation/navigation2_tutorials #85: errors in building the package  in ros2 rolling

Parikshit01, 2024-02-22:
i've tried to run the package in mentioned ros2 rolling but still getting some error  details of the terminal window is attached herewith as text file.

`Failed   <<< sam_bot_description [1.30s, exited with code 1]
Aborted  <<< nav2_straightline_planner [1.48s]                                          
Aborted  <<< nav2_sms_behavior [7.16s]                                  
Summary: 25 packages finished [24.4s]
  1 package failed: sam_bot_description
  2 packages aborted: nav2_sms_behavior nav2_straightline_planner
  13 packages had stderr output: examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher launch_testing_examples nav2_gps_waypoint_follower_demo nav2_sms_behavior `sam_bot_description`

[output.txt](https://github.com/ros-planning/navigation2_tutorials/files/14373972/output.txt)

elgarbe, 2024-02-23:
Do you need nav2_straightline_planner and nav2_sms_behavior packages? what do you try to use from this repo?
I remember removing those two package from the WP in order to get GPS WP tutorial, but I don't remember on which OS/ros version have to do that.
On iron all is working ok.

Parikshit01, 2024-02-23:
I was learning Nav2 from the tutorials for all the available features. Main for the time i am interested in the Navigating Using GPS Localization (https://navigation.ros.org/tutorials/docs/navigation2_with_gps.html) tutorial part implementation. I have no idea about the necessity of nav2_straightline_planner and nav2_sms_behavior packages. 
I will try with ros2 iron. 
Thanks a lot.

Parikshit01, 2024-02-26:
> Do you need nav2_straightline_planner and nav2_sms_behavior packages? what do you try to use from this repo? I remember removing those two package from the WP in order to get GPS WP tutorial, but I don't remember on which OS/ros version have to do that. On iron all is working ok.

Thanks. 
I tried with ros2 iron but  it also didn't ran successfully.

[output.txt](https://github.com/ros-planning/navigation2_tutorials/files/14400387/output.txt)

---
Finished <<< examples_rclpy_pointcloud_publisher [5.63s]
Starting >>> nav2_straightline_planner
Finished <<< nav2_costmap_filters_demo [2.71s]                                                      
Starting >>> sam_bot_description
--- stderr: sam_bot_description                                                                     
CMake Error at CMakeLists.txt:20 (find_package):
  By not providing "Findxacro.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "xacro", but
  CMake did not find one.

  Could not find a package configuration file provided by "xacro" with any of
  the following names:

xacroConfig.cmake
xacro-config.cmake

  Add the installation prefix of "xacro" to CMAKE_PREFIX_PATH or set
  "xacro_DIR" to a directory containing one of the above files.  If "xacro"
  provides a separate development package or SDK, be sure it has been
  installed.

Failed   <<< sam_bot_description [0.63s, exited with code 1]
Aborted  <<< launch_testing_examples [4.21s]            
Aborted  <<< nav2_gps_waypoint_follower_demo [3.90s]    
Aborted  <<< nav2_gradient_costmap_plugin [21.2s]       
Aborted  <<< nav2_sms_behavior [21.3s]                 
Aborted  <<< nav2_straightline_planner [26.5s]          
Aborted  <<< examples_rclcpp_wait_set [1min 50s]        
Aborted  <<< examples_rclcpp_minimal_subscriber [2min 9s]                                           

Summary: 20 packages finished [3min 8s]
  1 package failed: sam_bot_description
  7 packages aborted: examples_rclcpp_minimal_subscriber examples_rclcpp_wait_set launch_testing_examples nav2_gps_waypoint_follower_demo nav2_gradient_costmap_plugin nav2_sms_behavior nav2_straightline_planner
  11 packages had stderr output: examples_rclpy_executors examples_rclpy_guard_conditions examples_rclpy_minimal_action_client examples_rclpy_minimal_action_server examples_rclpy_minimal_client examples_rclpy_minimal_publisher examples_rclpy_minimal_service examples_rclpy_minimal_subscriber examples_rclpy_pointcloud_publisher nav2_sms_behavior sam_bot_description

Parikshit01, 2024-02-27:
> you need to run `rosdep install --from-paths src --ignore-src -r -y` on ws folder before `colcon build` in order to get all dependencies. From your error xacro package is missing.

Thanks a lot for your help. At least I'm able to run this with the ros iron running following one by one. 
sudo rosdep init
rosdep update
rosdep install --from-paths src --ignore-src -r -y

only I was having errors in nav2_sms_behavior package building. deleting that package, others are running successfully.

elgarbe, 2024-02-27:
No problem!
This new feature help a lot with real robots in outdoor environment. I'm working on a real robot right now and except for the poor precision of gps, everything is working right. I can navigate with LLA waypoints avoiding obstacles with Lidar data.
For the gps performance I will try with RTK.

closed 2024-11-18 by SteveMacenski