ros-navigation/navigation2_tutorials #86: WP "moves" on each field test on real robot

elgarbe, 2024-02-23:
Hi, I'm using nav2 GPS WP Logger and follower on a real robot on an outdoor environment.
First I drive the robot with a joystick and log each WP. Then I run nav2 on the robot and Rviz on a laptop. Unfortunately the robot doesn't follow the exact same WP. It's out for a few meters. I'm trying to understand why is this happening.
The first thing I've noticed is that as my robot moves the tf between map and odom drift. That is, odom frame is moving around from 0 to 2mts from map frame. In this point I wonder how it affect to navigation. I saw global cost-map is in map frame, but local cost-map y on odom frame. How is translated the GPS WP to nav2 WP? I know that from GPS to map y transformed using robot_localization, but then nav2 transform the WP to odom frame? Is there a topic that expose current WP goal?
I can add RTK corrections to my GPS, using it I could get GPS position with cm level, but that doesn't remove map to odom drift. Is there a way to remove that offset, moreover is it ok to try to remove that offset on robot_localization filter?

ZaBy10, 2024-05-02:
Hii @elgarbe , Could you please tell me how to run this package on a real robot . what are the changes that I need to make ?!

elgarbe, 2024-05-02:
You have to change sensor sources and comment out gazebo package.
You have to check what sensors your robot has and in topics name then make changes on robot_localization configuration file to use them. The same with lidar data.
It's a custom robot? what sensors do you have?

My problem reported in this post was related to GPS errors. I was able to solve it using RTK units.

Maybe you could open a new issue exposing your robot and your sensors and we may help there. I can share a repo with the code that I'm using.

kaleb236, 2024-09-26:
Hello @elgarbe , I'm trying to use it on a real robot too. Can you share your repo please?

elgarbe, 2024-09-26:
> Hello @elgarbe , I'm trying to use it on a real robot too. Can you share your repo please?

https://gitlab.com/chori_rover

closed 2024-11-18 by SteveMacenski