ros-navigation/navigation2_tutorials #87: Navigating Using GPS Localization - MapViz error

mattiadutto, 2024-03-21:
Hi, I tried to using the code of the Navigating Using GPS Localization and unfortunnaly out of the box it doesn't work (I know that I have to add the bing-api key). Searching on the web for a solution I saw that by replacing the mapviz configuration with this one (taken [here](https://github.com/swri-robotics/mapviz/issues/575#issuecomment-375669263)) it works:

```yaml

capture_directory: ~
fixed_frame: map
target_frame: map
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1000
window_height: 804
view_scale: 1
offset_x: 0
offset_y: 0
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: ""
      source: Stamen (terrain)
  - type: mapviz_plugins/point_click_publisher
    name: new display
    config:
      visible: true
      collapsed: false
      topic: clicked_point
      output_frame: wgs84
  - type: mapviz_plugins/tf_frame
    name: new display
    config:
      visible: true
      collapsed: false
      frame: base_link
      color: "#00ff00"
      draw_style: arrows
      position_tolerance: 0
      buffer_size: 1
      static_arrow_sizes: true
      arrow_size: 53
  - type: mapviz_plugins/navsat
    name: new display
    config:
      visible: true
      collapsed: false
      topic: /gps/fix
      color: "#55aaff"
      draw_style: points
      position_tolerance: 0
      buffer_size: 1
```

Could you please update the [gps configuration](https://github.com/ros-planning/navigation2_tutorials/blob/master/nav2_gps_waypoint_follower_demo/config/gps_wpf_demo.mvc) file?
Thanks.

System configuration: Ubuntu 22.04.3 LTS + ROS2 Humble + Navigation2 1.1.13

SteveMacenski, 2024-03-21:
Please submit a PR!

SteveMacenski, 2025-07-16:
Fixed in newest tutorial