ros-navigation/navigation2_tutorials #93: Issue with Loading Map Data from Mapviz Due to Bing Maps API Key Unavailability

jkoubs, 2024-07-17:
I'm having some issues with getting the map data from mapviz. The map does not load because I can't get a Bing Maps API key to display satellite pictures. Microsoft is no longer accepting new Bing Maps for Enterprise customers as of June 30, 2024 (details here: [Microsoft Bing Maps Discontinued Services](https://www.microsoft.com/en-us/maps/bing-maps/discontinued-services)).

Is there an alternative way to access this map data or a recommended workaround for this issue? Any help or suggestions would be greatly appreciated.

Thank you!

AhmedYasen97, 2024-07-18:
You can use offline Google Maps instead. I used [this](https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite) one, and it works fine with MapViz.

jkoubs, 2024-07-20:
Thanks @AhmedYasen97 This works for me.

closed 2024-07-20 by SteveMacenski