ros-navigation/navigation2_tutorials #105: Error adding inertia to robot base fram

raybietry, 2025-02-11:
When I ran the Writing the SDF section of the Nav2 tutorial for ROS2, Jazzy I was getting an error an error saying I was no allowed to add the inertia to the robot base. It needed to be added as a child link.
I had to removed the inertia macro from the base_link and added a new frame and it works.

The updated code is as follows:
    
    <link name='base_link'>
      <must_be_base_link>true</must_be_base_link>
      <visual name="base_link_visual">
        <geometry>
          <box><size>
            ${base_length} ${base_width} ${base_height}
          </size></box>
        </geometry>
        <material>
          <ambient>0 1 1 1</ambient>
          <diffuse>0 1 1 1</diffuse>
        </material>
      </visual>

      <collision name="base_link_collision">
        <geometry>
          <box><size>
            ${base_length} ${base_width} ${base_height}
          </size></box>
        </geometry>
      </collision>

	 
 	
    </link>

     

    <link name='base_inertia_joint' type='fixed'>
      <parent>base_link</parent>
      <child>base_inertia_link</child>
      <pose relative_to="base_link">0.0 0.0 0.0 0 0 0</pose>
 	 <xacro:box_inertia m="15" w="${base_width}" d="${base_length}" h="${base_height}"/>      
      
    </link>

SteveMacenski, 2025-02-11:
Wouldn't this already take care of it? https://github.com/ros-navigation/navigation2_tutorials/blob/rolling/sam_bot_description/src/description/sam_bot_description.sdf#L93

@Amronos did you run into any issues like this?

Amronos, 2025-02-13:
This comes as a warning from ``robot_state_publisher`` / ``kdl_parser``:

```
[robot_state_publisher-4] [WARN] [1739438120.126805114] [kdl_parser]: The root link base_link has an inertia specified in the URDF, but KDL does not support a root link with an inertia.  As a workaround, you can add an extra dummy link to your URDF.
```

This warning in my testing always comes, even when inertia is not specified for the root link (``base_link``). I am not sure what the exact issue here is and if it impacts anything.
This is related to https://github.com/ros/kdl_parser/issues/27.

SteveMacenski, 2025-02-13:
Should we move `<xacro:box_inertia m="15" w="${base_width}" d="${base_length}" h="${base_height}"/>` to the base footprint and remove the `<xacro:box_inertia m="0" w="0" d="0" h="0"/>` from the base footprint? That would remove any inertia added to the base link

Amronos, 2025-02-14:
> I assume you tried that though from your response (or did you have some inertia tag?)

I have tried that but the warning still appears (for SDF). Making similar changes in the URDF and using that instead of the SDF prevents the warning from coming. So I suppose ``sdformat_urdf``/``robot_state_publisher`` adds an inertia tag to all links.

I guess we can make this change in both URDF and SDF with a note in the docs that this warning will appear while using SDF but can be safely ignored.

SteveMacenski, 2025-02-14:
Ah got it - it might just be a quirk of the conversion

PR #107 (2025-03-04): Shift inertia to base_footprint
Fixes #105.

Corresponding documentation update: https://github.com/ros-navigation/docs.nav2.org/pull/646.
closed 2025-03-04 by SteveMacenski