ros-navigation/navigation2_tutorials #106: Keepout filter upon launching does not launch the nav2 server at all

Aishkrish18, 2025-02-18:
Hi ,
I wanted to try the Keepout filter integrated with a custom robot. Upon exactly following the steps as in the tutorial, it always gets stuck at this point :
ros2 launch nav2_costmap_filters_demo costmap_filter_info.launch.py params_file:=/home/neobotix/tutorials/src/navigation2_tutorials/nav2_costmap_filters_demo/params/keepout_params.yaml mask:=/home/neobotix/maps/keepout_mask.yaml
[INFO] [launch]: All log files can be found below /home/neobotix/.ros/log/2025-02-17-14-13-06-475444-blagg-42189
[INFO] [launch]: Default logging verbosity is set to INFO

It gets stuck here. I run the command on the humble branch. I am not quite sure what this issue could be. Just to test, i use the same Keepout mask and the params file as in the repository. Could you help me out?

Regards,
Aishwarya

AnOrdinaryUsser, 2025-02-28:
Hello,

I am also working on a robot built from scratch, and we are facing a similar issue. We can navigate with the robot and publish two maps:

The original map that we initially created.
A modified version of the map, where we have restricted the robot's movement to a designated corridor.
We followed the tutorial and cloned the repository into a ROS workspace, modifying the keepout_params.yaml file to use our modified map as the filter. However, despite working with the ROS Humble version, we have not been able to achieve the desired outcome.

Below is the output we get when running the command:

`~/ros2_ws$ ros2 launch nav2_costmap_filters_demo costmap_filter_info.launch.py params_file:=/home/david/ros2_ws/src/navigation2_tutorials/nav2_costmap_filters_demo/params/keepout_params.yaml mask:=/home/david/ros2_ws/src/navigation2_tutorials/nav2_costmap_filters_demo/maps/mapanave-lines.yaml use_composition:=True
[INFO] [launch]: All log files can be found below /home/david/.ros/log/2025-02-28-11-06-21-722047-david22-6216
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/filter_mask_server' in container '/nav2_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/costmap_filter_info_server' in container '/nav2_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/lifecycle_manager_costmap_filters' in container '/nav2_container'`

Have you managed to solve it somehow?

TZECHIN6, 2025-04-07:
@Aishkrish18 I am facing the same issue when using it with `use_composition:=True`. It will get stuck and no node (the nav2_container in here) is being spawned.

However if it using standalone method, it just work....

I am wondering is it this line causing the bug:
```
                condition=IfCondition(PythonExpression(["'",
                                                        LaunchConfiguration('namespace'),
                                                        "' != ''",])),
```

muhidabid, 2025-04-16:
I am facing the same issue; however, once I run `ros2 launch nav2_bringup tb3_simulation_launch.py` in another terminal, I get this:

```
# ros2 launch nav2_costmap_filters_demo costmap_filter_info.launch.py params_file:=`pwd`/src/navigation2_tutorials/nav2_costmap_filters_demo/params/keepout_params.yaml mask:=`pwd`/src/navigation2_tutorials/nav2_costmap_filters_demo/maps/keepout_mask.yaml use_composition:=True
[INFO] [launch]: All log files can be found below /root/.ros/log/2025-04-16-19-56-22-312365-ROGUE-PC-483
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/filter_mask_server' in container '/nav2_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/costmap_filter_info_server' in container '/nav2_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/lifecycle_manager_costmap_filters' in container '/nav2_container'
```

I am not sure if this is correct since the `keepout_mask.pgm` doesn't load in the RViz simulation. The simulation works fine though, just without the filter.

TZECHIN6, 2025-05-30:
I am able to use the keepout filter, but just not in compostable node launch. Might be you can try this first at least, to make your program work?

TZECHIN6, 2025-05-30:
Basically just follow the tutorial, and use standalone node to launch the keepout filter. That would work. Here is a example https://github.com/ros-navigation/navigation2_tutorials/tree/rolling/nav2_costmap_filters_demo

Make sure you have all the map and param ready.

If someone knows how to make the composition node works, please give me a post. Great thanks.

TZECHIN6, 2025-05-30:
Yes. Even though no error shown using composite node, there has no topic or server is up. I guess somehow it's not boot up in the lifecycle. But the standalone works. You can comment it out if you want to ensure launching from standalone node.

lyysl, 2025-05-30:
thx...it shows the mask in rviz2. But sometimes it has status:Warn in rviz2, and does not show anything. When I launch the launch file again, the image shows up again.

TZECHIN6, 2025-05-31:
I am not so sure what you mean? Are u using with nav2 or just showing the map? Could u upload a screencap.

lyysl, 2025-05-31:
I want to use nav2 eventually but I just want to test if this nav2_costmap_filters_demo works first.
The image is just showing a part of the turtlebot3 world map image in rviz2. If there is a full code for the navigation with keepout filter like [navigation2_with_keepout_filter](https://docs.nav2.org/tutorials/docs/navigation2_with_keepout_filter.html) would be great.
Thank you.

lyysl, 2025-06-02:
I could run the code now. I need to run with both the gazebo and the navigation as in the [KeepoutFilter demo youtube tutorial](https://www.youtube.com/watch?v=iVteYQR5IYQ). And the `costmap_filter_info.launch.py` is changed without using composite node.

```
ros2 launch nav2_costmap_filters_demo costmap_filter_info.launch.py

export GAZEBO_MODEL_PATH=$GAZEBO_MODEL_PATH:~/turtlebot3_ws/src/turtlebot3/turtlebot3_simulations/turtlebot3_gazebo/models
export TURTLEBOT3_MODEL=waffle_pi
ros2 launch turtlebot3_gazebo turtlebot3_world.launch.py

# keepout filter plugins are added in the nav2_params.yaml
ros2 launch nav2_bringup tb3_simulation_launch.py use_sim_time:=True autostart:=True map:=your_path/map.yaml
```
Thank you.

lyysl, 2025-06-10:
I have another problem about using multi robot for keepout filter. I am trying to use different robots in the same map.
But I want to use different keepout regions for different robots. I tried to use different map servers, but it seems I can only use one map server at a time. Are there any ways to do it for multi robots? Thank you.