ros-navigation/navigation2_tutorials #108: Is ‚ Navigating Using GPS Localization‘ Limited to map size?

nxpwilsch, 2025-02-26:
I tried the Navigating Using GPS Localization demo and goals that are away more than 500m have been rejected because out of map? Is this a bug or do I have to increase my map size to  e.g. 1km x 1km if distance to goal is 1km away? If yes, it seems that GPS coordinates not really make sense…

SteveMacenski, 2025-03-04:
Please ask on Robot Stack Exchange, this is the wrong place for configuration advice / help

closed 2025-03-04 by SteveMacenski

ghhhgfgh, 2025-11-07:
I'm facing the same issue. When the global map is set too small, navigation can only work within its boundaries. For example, with a 100x100 global map, it's basically limited to that square area, which is far from sufficient for outdoor environments. However, if I increase the costmap size, performance issues become significantly worse.