ros-navigation/navigation2_tutorials #110: use_odometry_yaw: true

Baominchao, 2025-03-10:
I learn this website:https://docs.nav2.org/tutorials/docs/navigation2_with_gps.html
The article mentioned：“in this tutorial we assume the robot is equipped with an IMU that can accurately measure absolute orientation following the ENU convention, meaning it will output zero yaw when facing east and +90 degrees when facing north.”
configuration for the navsat_transform node:
navsat_transform:
  ros__parameters:
    frequency: 30.0
    delay: 3.0
    magnetic_declination_radians: 0.0
    yaw_offset: 0.0
    zero_altitude: true
    broadcast_utm_transform: true
    publish_filtered_gps: true
    use_odometry_yaw: true
    wait_for_datum: false
    # datum: [38.161491, -122.4546443, 0.0] # pre-set datum if needed, [lat, lon, yaw]
I have a question about  use_odometry_yaw: true. Why do we still use the yam of the odometer? Isn't there already an IMU heading angle with absolute direction?

SteveMacenski, 2025-03-10:
Please review the documentation on robot localization - this is a R_L parameter

Baominchao, 2025-03-11:
Sorry, I know it's the R_L parameter, but I wanted to ask the author of this demo, which caused trouble for you.

closed 2025-03-11 by SteveMacenski