ros-navigation/navigation2_tutorials #124: Tutorial sources for kilted

timkoers, 2025-09-02:
Hi!
I'd like to use the tutorial as a reference for the implementation of plugins on ROS2 Kilted, but there are none available.

Could you port them to kilted?

timkoers, 2025-09-03:
When compiling the rolling branch, I get the following errors

```bash
--- stderr: nav2_pure_pursuit_controller
CMake Error at CMakeLists.txt:14 (find_package):
  By not providing "Findnav2_ros_common.cmake" in CMAKE_MODULE_PATH this
  project has asked CMake to find a package configuration file provided by
  "nav2_ros_common", but CMake did not find one.

  Could not find a package configuration file provided by "nav2_ros_common"
  with any of the following names:

    nav2_ros_commonConfig.cmake
    nav2_ros_common-config.cmake

  Add the installation prefix of "nav2_ros_common" to CMAKE_PREFIX_PATH or
  set "nav2_ros_common_DIR" to a directory containing one of the above files.
  If "nav2_ros_common" provides a separate development package or SDK, be
  sure it has been installed.

---
Failed   <<< nav2_pure_pursuit_controller [20.7s, exited with code 1]
```

Am I missing something? I've installed `ros-kilted-nav2-common`, but I can't find the `ros-kilted-nav2-ros-common` package

SteveMacenski, 2025-09-03:
Try this out: https://github.com/ros-navigation/navigation2_tutorials/tree/kilted

timkoers, 2025-09-03:
How strange, I couldn't previously find any kilted branch whatsoever.

Anyways, the kilted branch gives the following output, please note the warnings and errors

```bash
--- stderr: nav2_gradient_costmap_plugin
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(nav2_gradient_costmap_plugin_core PUBLIC
      nav2_costmap_2d::filters
      nav2_costmap_2d::layers
      nav2_costmap_2d::nav2_costmap_2d_client
      nav2_costmap_2d::nav2_costmap_2d_core
      pluginlib::pluginlib
      rclcpp::rclcpp
    )

Call Stack (most recent call first):
  CMakeLists.txt:51 (ament_target_dependencies)

---
Finished <<< nav2_gradient_costmap_plugin [32.0s]
--- stderr: nav2_straightline_planner
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(nav2_straightline_planner_plugin PUBLIC
      ${builtin_interfaces_TARGETS}
      ${geometry_msgs_TARGETS}
      ${nav2_msgs_TARGETS}
      ${nav_msgs_TARGETS}
      ${std_msgs_TARGETS}
      ${visualization_msgs_TARGETS}
      nav2_core::nav2_core
      nav2_costmap_2d::filters
      nav2_costmap_2d::layers
      nav2_costmap_2d::nav2_costmap_2d_client
      nav2_costmap_2d::nav2_costmap_2d_core
      nav2_util::nav2_util_core
      pluginlib::pluginlib
      rclcpp::rclcpp
      rclcpp_action::rclcpp_action
      rclcpp_lifecycle::rclcpp_lifecycle
      tf2_ros::static_transform_broadcaster_node
      tf2_ros::tf2_ros
    )

Call Stack (most recent call first):
  CMakeLists.txt:55 (ament_target_dependencies)

---
Finished <<< nav2_straightline_planner [37.3s]
--- stderr: nav2_pure_pursuit_controller
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(nav2_pure_pursuit_controller PUBLIC
      ${geometry_msgs_TARGETS}
      ${nav_msgs_TARGETS}
      nav2_core::nav2_core
      nav2_costmap_2d::filters
      nav2_costmap_2d::layers
      nav2_costmap_2d::nav2_costmap_2d_client
      nav2_costmap_2d::nav2_costmap_2d_core
      nav2_util::nav2_util_core
      pluginlib::pluginlib
[... 66 lines cut ...]
CMake Deprecation Warning at /opt/ros/kilted/share/ament_cmake_target_dependencies/cmake/ament_target_dependencies.cmake:87 (message):
  ament_target_dependencies() is deprecated.  Use target_link_libraries()
  with modern CMake targets instead.  Try replacing this call with:

    target_link_libraries(test_path_utils PUBLIC
      ${geometry_msgs_TARGETS}
      ${nav_msgs_TARGETS}
      tf2_geometry_msgs::tf2_geometry_msgs
    )

Call Stack (most recent call first):
  test/CMakeLists.txt:17 (ament_target_dependencies)

/mnt/c/Users/Tim/Documents/Projecten/BulbPlaint/BulbPlacer/src/highlevel/navigation/vector_pursuit_controller/src/vector_pursuit_controller.cpp:27:10: fatal error: nav2_core/exceptions.hpp: No such file or directory
   27 | #include "nav2_core/exceptions.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/vector_pursuit_controller.dir/build.make:76: CMakeFiles/vector_pursuit_controller.dir/src/vector_pursuit_controller.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:174: CMakeFiles/vector_pursuit_controller.dir/all] Error 2
gmake[1]: *** Waiting for unfinished jobs....
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< vector_pursuit_controller [39.2s, exited with code 2]
Aborted  <<< nav2_sms_behavior [58.8s]
```

closed 2025-09-03 by timkoers